deviceplug/btleplug #251: `characteristics()` and `services()` of `Peripheral` returning a `HashMap` instead of a `BTreeSet`

hansmbakker, 2022-02-20:
**Feature Description**

Would it be possible to have `characteristics()` and `services()` of `Peripheral` return a `HashMap` instead of a `BTreeSet`?

That way, looping over the characteristics or calling `iter().find()` to find the characteristic you need is not needed anymore as below:
https://github.com/deviceplug/btleplug/blob/76fd7956352fc59e35dc4e27d9de285df0554e43/examples/lights.rs#L63-L66

Instead you could do something like
```rust
let cmd_char = light.characteristics()[&LIGHT_CHARACTERISTIC_UUID];
```

rib, 2022-06-07:
One thing to consider here is that bluetooth le peripherals may advertise multiple instances of a service or characteristic with the same Uuid (differentiated by their underlying AT handle) which couldn't be supported by btleplug if they were only exposed via a  `HashMap`, indexed by uuid.

Their on-device order may also be significant, which would be lost with a `HashMap`.

thomaseizinger, 2022-07-20:
> ```rust
> let cmd_char = light.characteristics()[&LIGHT_CHARACTERISTIC_UUID];
> ```

If it is just about API ergonomics, then a new type around the returned `BTreeSet` plus an imlementation of [`Index`](https://doc.rust-lang.org/std/ops/trait.Index.html) would also allow this API.