deviceplug/btleplug #257: BlueR

hansmbakker, 2022-03-19:
This is not a feature request, but I came across https://github.com/bluez/bluer and thought you might find it interesting. You might already know this library or see no use for it - feel free to close this issue.

It is described as the official Bluez Rust library. I don't know its maturity compared to the bluez-async crate that btleplug is using - at least when looking at its github stars BlueR seems to be more popular.

qwandor, 2022-03-21:
Yes, `bluer` is more recent than `bluez-async`, it didn't exist when I wrote `bluez-async` or used it for `btleplug`. `bluer` does have several features which `bluez-async` doesn't, such as Bluetooth Classic and publishing GATT services, but `btleplug` doesn't have any plan to support these features so it doesn't make much difference to us. I haven't looked at `bluer` in very much depth, but from what I remember their approach to event streams is a bit different and would be a bit more work to fit into `btleplug`'s model. I don't have the time or motivation to investigate further, as `bluez-async` works pretty well, but if you want to try prototyping how it would look to use `bluer` instead then you could do.

qdot, 2022-09-19:
Gonna close this as I've added Bluer to our README in case people want alternatives.

closed 2022-09-19 by qdot