deviceplug/btleplug #260: Can't connect peripheral on

dongjian, 2022-06-30:
**Describe the bug**
A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
A clear and concise description of what actually happened.

**Additional context**
Add any other context about the problem here.

qdot, 2022-06-30:
This is gonna put us in direct conflict with the bug where having it set to cached caused update issues on some devices, so I'm not sure there's an easy fix here. https://github.com/deviceplug/btleplug/issues/155

dongjian, 2022-07-01:
Thanks for you reply. and another question need help as mentioned above. I found on windows device need to connect to the system first . then you can connect it correctly. or it will be err  { code: 0x8000FFFF, message: catastrophic failure
 } . Is there any way can make the device auto connect to the system when call ```peripheral.connect().awit ``` on windows ?