deviceplug/btleplug #273: "Random" panic in CBPeripheral::confirm_disconnect, likely when device loses connection

mmastrac, 2022-10-22:
**Describe the bug**

When sending a set of status update requests to a BLE device, the code eventually panics deep within the `CoreBluetooth` interface. I can't reproduce it reliably, but running long enough will result in this crash.

**Expected behavior**

No panic, error message propagates up.

**Actual behavior**

Unexpected panic 

**Additional context**

It's difficult to say exactly what causes this, but in this case I'm simply reading/writing to a characteristic.

```
thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.10.1/src/corebluetooth/internal.rs:238:14
stack backtrace:
   0: rust_begin_unwind
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/std/src/panicking.rs:584:5
   1: core::panicking::panic_fmt
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/core/src/panicking.rs:142:14
   2: core::panicking::panic
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/core/src/panicking.rs:48:5
   3: core::option::Option<T>::unwrap
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/core/src/option.rs:755:21
   4: btleplug::corebluetooth::internal::CBPeripheral::confirm_disconnect
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.10.1/src/corebluetooth/internal.rs:236:9
   5: btleplug::corebluetooth::internal::CoreBluetoothInternal::on_peripheral_disconnect::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.10.1/src/corebluetooth/internal.rs:506:13
   6: <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/core/src/future/mod.rs:91:19
   7: btleplug::corebluetooth::internal::CoreBluetoothInternal::wait_for_message::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.10.1/src/corebluetooth/internal.rs:774:71
   8: <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/core/src/future/mod.rs:91:19
   9: btleplug::corebluetooth::internal::run_corebluetooth_thread::{{closure}}::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.10.1/src/corebluetooth/internal.rs:909:39
  10: <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/core/src/future/mod.rs:91:19
  11: <core::pin::Pin<P> as core::future::future::Future>::poll
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/core/src/future/future.rs:124:9
  12: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}::{{closure}}::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:525:48
  13: tokio::coop::with_budget::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/coop.rs:102:9
  14: std::thread::local::LocalKey<T>::try_with
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/std/src/thread/local.rs:445:16
  15: std::thread::local::LocalKey<T>::with
             at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library/std/src/thread/local.rs:421:9
  16: tokio::coop::with_budget
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/coop.rs:95:5
  17: tokio::coop::budget
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/coop.rs:72:5
  18: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:525:25
  19: tokio::runtime::scheduler::current_thread::Context::enter
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:349:19
  20: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:524:36
  21: tokio::runtime::scheduler::current_thread::CoreGuard::enter::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:595:57
  22: tokio::macros::scoped_tls::ScopedKey<T>::set
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/macros/scoped_tls.rs:61:9
  23: tokio::runtime::scheduler::current_thread::CoreGuard::enter
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:595:27
  24: tokio::runtime::scheduler::current_thread::CoreGuard::block_on
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:515:19
  25: tokio::runtime::scheduler::current_thread::CurrentThread::block_on
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/scheduler/current_thread.rs:161:24
  26: tokio::runtime::Runtime::block_on
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.21.2/src/runtime/mod.rs:490:46
  27: btleplug::corebluetooth::internal::run_corebluetooth_thread::{{closure}}
             at /Users/matt/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.10.1/src/corebluetooth/internal.rs:906:9
```

mmastrac, 2022-10-23:
This might be a dupe of #271 if the stack trace is being hidden in that case.

mmastrac, 2022-11-02:
@qdot This is fixed in 0.10.2, confirmed.

qdot, 2022-11-02:
Closing, thanks for confirming @mmastrac !

closed 2022-11-02 by qdot