deviceplug/btleplug #275: Make uuid in PeripheralId `pub`

mmastrac, 2022-10-25:
**Feature Description**

The `PeripheralId` can be converted `from` a `Uuid`, but there's no easy way to get the internal `uuid` property. This is useful for fast reconnection, as you can just repeatedly ask the adapter for a peripheral with a given `PeripheralId` rather than wait for it to be discovered or continuously scan all discovered devices.

qdot, 2022-10-25:
Idea already rejected in #266, followup landed in #268, will go out with next point release, hopefully this week. Closing as dupe.

closed 2022-10-25 by qdot

qwandor, 2022-10-25:
If you just want to get a `Peripheral` which you previously discovered, you can call `Central::peripheral` with the `PeripheralId`, there's no need to get the internal details out of it. The `Uuid` is only on MacOS anyway, it's different on Windows or Linux.