deviceplug/btleplug #276: RSSI and tx_power_level are None on OSX (corebluetooth)

Yohannfra, 2022-10-29:
**Describe the bug**
The **rssi** and **tx_power_level** properties is set to None in the code. in *src/corebluetooth/peripheral.rs*
```rust
        let properties = Mutex::from(PeripheralProperties {
            address: BDAddr::default(),
            address_type: None,
            local_name,
            tx_power_level: None,
            rssi: None,
            manufacturer_data: HashMap::new(),
            service_data: HashMap::new(),
            services: Vec::new(),
        });
```

**Expected behavior**
Actual rssi and tx_power_level.

**Actual behavior**
No rssi and tx_power_level

Thanks you

ztroop, 2024-03-02:
I'm currently developing a tool that depends on this library. `0.11.5`, on a Mac (M2). I notice I only get `Option -> None` with `tx_power_level` in every case. It would seem this is still an issue. However, I do see values for `rssi`

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because CoreBluetooth now populates RSSI from advertisements and supports RSSI reads, and extracts tx power when the platform provides it. These values remain optional when the peripheral or operating system does not expose them.

closed 2026-08-30 by qdot