deviceplug/btleplug #277: Turning off OSX bluetooth adapter results in undetectable disconnect

mmastrac, 2022-11-02:
**Describe the bug**

In 0.10.1, turning off the BT adapter on a MacBook 2019 Intel device would cause a disconnection event 100% of the time. It would print `Event receiver died, breaking out of corebluetooth device loop` multiple times, but disconnects would be correctly received. 

In 0.10.2, turning off the BT adapter prints `Event receiver died, breaking out of corebluetooth device loop` and no disconnect events are received.

**Expected behavior**

Turning off the bluetooth adapter should send a disconnect event for all peripherals.

**Actual behavior**

Nothing happens, and the device does not send a disconnection event. Something may also be blocking the tokio event loop, as a `select!` that includes the BT device and a timeout stops sending the timeout.

**Additional context**

0.10.1 _sort of_ worked, but 0.10.2 is definitely not working.

qdot, 2022-11-02:
Ooof. Feels like the weight of our crappy macOS kludge may be getting the better of us. Will take a look, thanks for letting me know.

mmastrac, 2022-11-03:
No worries. I've worked around this for now using a non-tokio thread with a watchdog that detects a wedged future and it seems to be working.

I'm currently putting the final polish on my project that makes use of it - I can definitely dig into the OSX code and get my hands dirty.

Apologies for reporting so many bugs and forgetting to mention what an awesome library this is. It's pretty much the core part of my latest project (https://github.com/mmastrac/longshot/). 😀

bxb100, 2023-12-25:
Currently, I use a timeout before `write`

code like this:

```rust

   // edge case: https://github.com/deviceplug/btleplug/issues/277
    tokio::select! {
        _ = time::sleep(Duration::from_secs(2)) => {
            log::error!(target: "call_printer", "printer timeout");
            return Err(anyhow!("printer timeout"));
        }
        // connect to the device
        _ = printer.is_connected() => {
            log::debug!("connected to printer");
        }
    }

```

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because CoreBluetooth now handles adapter power-off transitions by emitting disconnect events, cancelling pending operations, and clearing affected peripheral state.

closed 2026-08-30 by qdot