deviceplug/btleplug #287: Android: Fatal signal 6 (SIGABRT) when getting the adapter

Avani-A-S, 2022-12-26:
**Describe the bug**
For Android platform, getting this error while getting the adapter.
`Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 16382 (rustapplication), pid 16382 (rustapplication)`

```
    let manager = Manager::new().await.unwrap();

    log::info!("get the first bluetooth adapter");
    // get the first bluetooth adapter
    let adapters = manager.adapters().await?;
```

Taking the code reference from https://docs.rs/btleplug/0.10.4/btleplug/. This work fine on windows. When trying to cross compile for android, creating SIGABRT error.

Provided more details https://users.rust-lang.org/t/btleplug-on-android-app-issue/86339.

qdot, 2022-12-26:
How are you building the android system? Did you build and bring in the droidplug classes from btleplug too? I'm wondering if this is a missing java symbol or something.

Avani-A-S, 2022-12-27:
I have followed steps in this blog. I created the app with same steps and then added the btleplug too.
https://blog.devgenius.io/integrating-rust-with-android-development-ef341c2f9cca

This is my Cargo.toml file.
[image]

qdot, 2022-12-27:
That blog has nothing to do with our project. Did you follow the extra steps outlined in our README? https://github.com/deviceplug/btleplug#android

Avani-A-S, 2022-12-27:
First, I started implementation as per README only. Faced same error for adapter so I googled some more and found devgenius blog. Let me try again as per README and I'll share the issue with code.

Avani-A-S, 2022-12-27:
Will you be able to list out set up steps in detail. Not sure how to build java portion of btleplug or jni-util-rs and how to connect it with my android app from the steps.

I am new with rust and android both. For local set up I need to checkout both this and add to my project?
https://github.com/deviceplug/jni-utils-rs, classcache branch
https://github.com/deviceplug/btleplug, android-tokio branch

Avani-A-S, 2022-12-27:
Added the base code here. You might look into it provide some pointers.
https://github.com/Avani-A-S/bledemoapp

Avani-A-S, 2022-12-30:
@qdot Followed these steps as per README and yet facing same issue. Pushed latest changes here https://github.com/Avani-A-S/bledemoapp.

Steps followed:

1. Checked out jni-utils-rs and btleplug locally. 
2. Opened jni-utils-rs in Android Studio and build libs at path "jni-utils-rs\target\debug\java\libs" with following commands.
> $ cargo build
> $ cd java
> $ ./gradlew build
3. Opened btleplug/src/droidplug/java in Android Studio. Updated build.gradle for above generated lib dependencies.
> dependencies {
>     compile fileTree(dir: '/jni-utils-rs/target/debug/java/libs', include: ['*.jar'])
> }
4. Build -> Rebuild Project. AAR generated at 'btleplug\src\droidplug\java\build\outputs\aar\droidplug-debug.aar'
5. Placed above generated .aar at my android app location 'android\app\libs'
6. Added .aar file dependencies in build.gradle.
[image]

7. Run the app, clicked on scan button and still facing same error when trying to get the adapter.
[image]

Oppzippy, 2023-01-03:
I was getting `Fatal signal 6 (SIGABRT)` as well, and in my case it, turned out to be not calling `btleplug::platform::init` (panic was here I think: https://github.com/deviceplug/btleplug/blob/master/src/droidplug/mod.rs#L20). Building with debug symbols made it easy to find.

I don't think `btleplug::platform::init` is mentioned anywhere in the readme or docs, so that should probably be added.

Avani-A-S, 2023-01-03:
Your issue resolved with `btleplug::platform::init`? Can you tell me where to use it.

Oppzippy, 2023-01-03:
It's a little outdated but see the code snippet here https://github.com/deviceplug/btleplug/issues/8#issuecomment-1107362866. It doesn't really matter where you put it, it just needs to be called before you do anything with btleplug. So putting it in an init function that you call manually [like this](https://github.com/Oppzippy/OpenSCQ30/blob/46854cc32d39e100db45bb230d6f7c94c9fd02b4/android/src/btleplug_init.rs#L6-L13) is fine too.

Avani-A-S, 2023-01-04:
Yes. I tried that but facing error `expected struct jni::wrapper::jnienv::JNIEnv, found struct JNIEnv`.
[image]

Facing the same issue with flutter as well so reported https://github.com/trobanga/btleplugtest/issues/4.

Not sure if this is the issue related to versions of libraries I am using -
jni-utils - 0.1.0
btleplug - 0.10.4
jni - 0.20.0

Oppzippy, 2023-01-04:
jni should be [0.19](https://github.com/deviceplug/btleplug/blob/d6d8ba5486eba5d78e5f5df6ddd632036b14f4de/Cargo.toml#L44)

Avani-A-S, 2023-01-06:
@Oppzippy Thanks a lot! It finally worked with jni 0.19.0 and by adding btleplug::platform::init.

closed 2023-01-06 by Avani-A-S