deviceplug/btleplug #289: Devices sending multiple advertisements get filtered on bluez/linux

micolous, 2023-01-09:
**Describe the bug**

Devices sending multiple types of advertisements (manufacturer data, service data) get de-duplicated by `bluez` by default, so you only see _new_ advertisement types from a device around 10% of the time.

This is specifically an issue with scanning Android and iOS devices, because they send out advertisements pretty much constantly, so it misses "new" things.

An easy way to trigger this is by using [caBLE (hybrid FIDO/WebAuthn authenticators)](https://android-developers.googleblog.com/2022/10/bringing-passkeys-to-android-and-chrome.html); which you can do on Android devices with a recent version of Chrome and Google Play Services (October 2022), or with iOS 16 or later[^ios] -- scanning such a QR code will cause it to start advertising service data payloads with UUID `0xfde2` or `0xfff9`.

**Steps to reproduce:**

1. Run the `event_driven_discovery` example on Linux / bluez

2. Scan this QR code[^url] with a device running iOS 16 or later[^ios] or [Android with a version of Chrome and Google Play Services from October 2022 or later](https://developers.google.com/identity/passkeys/supported-environments):

   [image]

   This will start the "make credential" flow on the device, and cause it to start sending new service data advertisements with UUID `0xfde2` or `0xfff9`.  That should time out after about 60 seconds, because the next step of the protocol requires the initiator to take some data in that service data advertisement and use it to connect to a Websocket tunnel that the device is waiting on.

[^url]: FIDO:/196944614260737660800636836935388126324298063435663648500979208347122763132056627778028993444361869915189377361839923290058725964354330451710836675838722109321447142404
[^ios]: iOS 15 scans it, but doesn't work correctly.

**Expected behaviour**

`ServiceDataAdvertisement` events for `0000fde2-0000-1000-8000-00805f9b34fb` or `0000fff9-0000-1000-8000-00805f9b34fb` after the above QR code is scanned.

**Actual behaviour**

No `ServiceDataAdvertisement` events for those UUIDs (about 90% of the time).

**Additional context**

* Tested this with a fresh Debian 11.6.0 install, which has Linux 5.10.0-20-amd64, bluez 5.33-3.1, rust 1.66.0 (installed from rustup)

* Running the same `event_driven_discovery` example on macOS and Windows works correctly

* Setting `filter.duplicate_data = Some(true)` in `impl Central for Adapter::start_scan()` (in `src/bluez/adapter.rs`) _fixes_ the issue.

  This is despite [the documentation saying `DuplicateData` is set to `true` by default](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt?id=6e49216ad47dc8ffd9cbf2066d13fd7b59613c98#n112); even tracing `btleplug`'s DBus calls with `bustle` shows that it's not setting that field by default; and there's different behaviour.

* If the above QR code stops working (it has a timestamp in it, but clients don't appear to check it), you can get a new one by registering a new WebAuthn credential using Chrome or Safari, then selecting "authenticate with another device", which will give you a new one.  [There's a demo site here which will do it](https://webauthn.firstyear.id.au).

micolous, 2023-01-10:
Closing this because #290 was merged that fixes this issue.

micolous, 2023-01-10:
That's still on the `dev` branch, so I'll leave this open for now.

ckcr4lyf, 2023-04-12:
Thanks for the fix.

PR #290 (2023-01-09): Set `DuplicateData = true` for bluez
Fixes #289

This makes bluez discovery work like it does on macOS and Windows.
commit bb3212df (2023-01-09): Set `DuplicateData = true` for bluez. Fixes #289
commit 2bd4d150 (2023-04-11): Set `DuplicateData = true` for bluez. Fixes #289
closed 2023-04-13 by qdot