deviceplug/btleplug #291: Android: Failed to connect BLE Device on android app

Avani-A-S, 2023-01-11:
**Describe the bug**
Failed to connect BLE Device on android. 

**Expected behavior**
Should connect to BLE device from Android App. With same code able to connect with same BLE device from Windows console app.

**Actual behavior**
Catching Security Exception in `connect()` in `Peripheral.java` in `droidplug`.
Debugged and observed that `this.gatt` initialized successfully. Not sure why throwing exception. Not able to see `var4` value.
[image]

**Additional context**
Log
```
2023-01-11 14:05:47.715 17510-17510 ViewRootIm...nActivity] com.example.domain            I  ViewPostIme pointer 0
2023-01-11 14:05:47.715 17510-17510 MSHandlerLifeCycle      com.example.domain            I  isMultiSplitHandlerRequested: windowingMode=1 isFullscreen=true isPopOver=false isHidden=false skipActivityType=false isHandlerType=true this: DecorView@1869565[MainActivity]
2023-01-11 14:05:47.760 17510-17510 ViewRootIm...nActivity] com.example.domain            I  ViewPostIme pointer 1
2023-01-11 14:05:47.760 17510-17510 MSHandlerLifeCycle      com.example.domain            I  isMultiSplitHandlerRequested: windowingMode=1 isFullscreen=true isPopOver=false isHidden=false skipActivityType=false isHandlerType=true this: DecorView@1869565[MainActivity]
2023-01-11 14:05:47.761 17510-17510 AbsListView             com.example.domain            D  onTouchUp() mTouchMode : 0
2023-01-11 14:05:47.838 17510-17510 BluetoothAdapter        com.example.domain            I  STATE_ON
2023-01-11 14:05:47.838 17510-17510 BluetoothAdapter        com.example.domain            I  STATE_ON
2023-01-11 14:05:47.838 17510-17510 BluetoothAdapter        com.example.domain            I  STATE_ON
2023-01-11 14:05:47.838 17510-17510 BluetoothAdapter        com.example.domain            I  STATE_ON
2023-01-11 14:05:47.839 17510-17510 BluetoothLeScanner      com.example.domain            D  Start Scan with callback
2023-01-11 14:05:47.841 17510-17918 BluetoothLeScanner      com.example.domain            D  onScannerRegistered() - status=0 scannerId=11 mScannerId=0
2023-01-11 14:05:47.845  5475-6032  BtGatt.GattService      pid-5475                             E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.example.domain, scannerId: 11, reportDelayMillis=0
2023-01-11 14:06:17.846 17510-17510 Rust                    com.example.domain            I  rust_core::device: Connecting to peripheral "Mi Band 3".
2023-01-11 14:06:17.852 17510-17510 BluetoothAdapter        com.example.domain            I  STATE_ON
2023-01-11 14:06:17.858 17510-17510 BluetoothGatt           com.example.domain            D  connect() - device: D2:EF:19:EB:77:E0, auto: false
2023-01-11 14:06:17.859 17510-17510 BluetoothAdapter        com.example.domain            I  isSecureModeEnabled
2023-01-11 14:06:17.862 17510-17510 BluetoothGatt           com.example.domain            D  registerApp()
2023-01-11 14:06:17.863 17510-17510 BluetoothGatt           com.example.domain            D  registerApp() - UUID=b0b37ce8-e6df-4ea8-8553-935953a22422
2023-01-11 14:06:17.874 17510-17918 BluetoothGatt           com.example.domain            D  onClientRegistered() - status=0 clientIf=10

> **2023-01-11 14:06:17.967 17510-17510 libc                    com.example.domain            A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 17510 (ware.domain), pid 17510 (ware.domain)**

2023-01-11 14:06:18.943 18856-18856 DEBUG                   pid-18856                            A  Cmdline: com.example.domain
2023-01-11 14:06:18.943 18856-18856 DEBUG                   pid-18856                            A  pid: 17510, tid: 17510, name: ware.domain  >>> com.example.domain <<<
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #01 pc 0000000000117a38  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (std::sys::unix::abort_internal::hb78ca46b57303bb0+4)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #02 pc 0000000000115060  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (rust_panic+128)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #03 pc 0000000000114e68  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (std::panicking::rust_panic_with_hook::h4d877f67398ec700+664)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #04 pc 0000000000114b6c  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (std::panicking::begin_panic_handler::_$u7b$$u7b$closure$u7d$$u7d$::hb2bc281294582b1d+132)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #05 pc 000000000011352c  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (std::sys_common::backtrace::__rust_end_short_backtrace::h9024ef89f901fb94+28)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #06 pc 00000000001148fc  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (rust_begin_unwind+108)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #07 pc 000000000012da08  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (core::panicking::panic_fmt::hfab367bd32819f09+44)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #08 pc 000000000012da78  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (core::panicking::panic::he54c28971e117703+48)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #09 pc 00000000000b83c8  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (jni_utils::future::JFuture::from_env::h6fbbc1005371d0de+440)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #10 pc 00000000000aa3ac  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (_$LT$btleplug..droidplug..peripheral..Peripheral$u20$as$u20$btleplug..api..Peripheral$GT$::connect::_$u7b$$u7b$closure$u7d$$u7d$::hae5db3f55b867c17+248)
2023-01-11 14:06:18.944 18856-18856 DEBUG                   pid-18856                            A        #11 pc 0000000000071538  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (tokio::runtime::park::CachedParkThread::block_on::h1ddd960ab0d93a2d+404)
2023-01-11 14:06:18.945 18856-18856 DEBUG                   pid-18856                            A        #12 pc 0000000000065c4c  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (tokio::runtime::scheduler::multi_thread::MultiThread::block_on::h995ebe14cea7adf2+104)
2023-01-11 14:06:18.945 18856-18856 DEBUG                   pid-18856                            A        #13 pc 00000000000626c8  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (tokio::runtime::runtime::Runtime::block_on::h818a95d3c494076a+100)
2023-01-11 14:06:18.945 18856-18856 DEBUG                   pid-18856                            A        #14 pc 0000000000064b0c  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (rust_core::device::connect::h6af38a96272b5cb9+160)
2023-01-11 14:06:18.945 18856-18856 DEBUG                   pid-18856                            A        #15 pc 0000000000062d50  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (rust_core::core_connect::h64a058cc01dcd19e+44)
2023-01-11 14:06:18.945 18856-18856 DEBUG                   pid-18856                            A        #16 pc 0000000000060c60  /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk (Java_com_example_domain_MainActivity_connect+112)
2023-01-11 14:06:18.945 18856-18856 DEBUG                   pid-18856                            A        #19 pc 0000000000000c0e  [anon:dalvik-classes3.dex extracted in memory from /data/app/~~n4A2OyJW1QWzZkjolgRw2Q==/com.example.domain-XtBkcMpzxilZ3pf6nnYcDg==/base.apk!classes3. [... cut ...]
---------------------------- PROCESS ENDED (17510) for package com.example.domain ----------------------------
2023-01-11 14:06:19.103  5475-6032  BtGatt.GattService      pid-5475                             E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: com.example.domain, scannerId: 11, reportDelayMillis=0
2023-01-11 14:06:19.105  5475-6032  LeAppInfo               pid-5475                             E  updateScanStopInfo(), There is NO scan process with :                          com.example.domain|    Normal|  LowPower| NonFilter|      true|

```

onecalfman, 2023-01-16:
Have you granted the location permission to your app in the devices settings and set up your manifest according to the [android docs](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions)?

Avani-A-S, 2023-01-16:
Yes and yes.
https://github.com/Avani-A-S/bledemoapp/blob/main/android/app/src/main/AndroidManifest.xml

Hero9909, 2023-01-20:
after a quick look i think this might be related to your permission setup or the timing. as stated before the permission should be set up like on the android docs, but your manifest is using the legacy and current used permissions. if you target sdk 31, you should either remove legacy or tag them via the attribute to remove them.  if your code has nothing todo with the location you should also remove this permission. in my experience this behaviour is also influenced by the device itself. Still a last thing, if the device is not very performant on some (expected as you also target sdk23) the permissions take time to be granted. await the result before scanning.

Avani-A-S, 2023-01-31:
I tried by removing legacy and location permissions. Keeping BLUETOOTH_CONNECT, SCAN and ADVERTISE only. Still the same results. Devices are scanned and listed fine. On selecting the device, I am calling the scan method again, matching the name of peripheral and sending connection request. Does not seem like timing issue.

Avani-A-S, 2023-02-09:
Check out https://github.com/deviceplug/jni-utils-rs, class cache branch and add it as local dependency resolved the issue with future.