deviceplug/btleplug #293: Linker error since rustc version 1.65.0 for iOS target

Arwed-M, 2023-01-19:
**Describe the bug**
When compiling a project for the target ```aarch64-apple-ios``` or ```x86_64-apple-ios``` featuring this library with rustc version ```1.65.0``` or greater I get the following error:

```
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-Wl,

[... following an endless list of directories ...]

= note: ld: framework not found AppKit
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

**Current work-around**
Compiling with rustc version 1.64.0 works just fine.

jhonboy121, 2023-04-19:
Should be fixed with https://github.com/deviceplug/btleplug/pull/302

Bradley-Cooley, 2024-12-04:
Were you using nix by any chance? I'm doing so on darwin with devenv and found that the issue was fixed by including the following packages in my dev shell:
1. pkgs.darwin.apple_sdk.frameworks.AppKit
2. pkgs.darwin.apple_sdk.frameworks.CoreBluetooth

Arwed-M, 2024-12-09:
> Were you using nix by any chance?

Good to know! No I use homebrew so I can't confirm ...

qdot, 2026-08-31:
Closing this out because we're like 30-some versions of rust down the line now, and things have been compiling fine for me. That said, if this does continue to be an issue, please reopen or refile.

closed 2026-08-31 by qdot