deviceplug/btleplug #300: Windows: disconnect is not working

jeffer80, 2023-03-06:
**Describe the bug**
on Windows, when calling the peripheral.disconnect() to end the connection with a BLE device, it executes successfully, 
but the connection still existed until I exit the main process.
I can receive the DeviceDisconnected event, and everything looks OK, except the BLE device is still connected to my PC.

qdot, 2023-03-06:
What version of windows are you on?

jeffer80, 2023-03-07:
> 

I have tested it on win10 and win11

jeffer80, 2023-03-07:
> What version of windows are you on?

I have tested it on win10 and win11

timmyhadwen, 2023-03-27:
Hi @qdot and @jeffer80.

We are experiencing the same behaviour on Windows 10 22H2 (19045.2728). We are going to attempt on some other versions and will report back with the results.

Cheers
Tim

oletf, 2023-05-19:
Greetings, 

experiencing the same issue on Windows 10 22H2 (19045.2965)

Any news on this ?

Thanks forward

qdot, 2026-10-10:
This is now over 3 years old, closing. If there's still issues, please reopen or open a new bug.

PR #329 (2023-08-24): Fix disconnect when using the winrt backend.
Fixes #300 

You must explicitly disconnect from each service before disconnecting from the device. Otherwise, Windows does not seem to fully disconnect the device internally.

See https://stackoverflow.com/questions/39599252/windows-ble-uwp-disconnect for similar issues.

I did this by simply adding a list of services to the `BLEDevice` when `discover_services` is called, allowing us to `Close` them when the device is dropped. This appears to work in my use-case, but I'd be interested to see if it's a robust enough solution for other folks.
closed 2026-10-10 by qdot