deviceplug/btleplug #301: Peripheral.Connect halts on windows 11

mrhatman, 2023-03-31:
**Describe the bug**
When attempting to connect to a BLE device on Windows 11 it stalls out forever. 

**Expected behavior**
peripheral.connect().await should connect or return an error but on windows 11 it does not.

**Additional context**
The exact same codebase runs on multiple Windows 10 laptops and fails across multiple windows 11 laptops( across multiple companies too)

I  have tried updating the windows crate to version 0.46.0 (latest) and I tried pull number #269 with no luck on either device.

jesterfraud, 2023-10-18:
I've encountered a deadlock on `Peripheral.connect()` on Windows 11 as well. It doesn't happen on initial connection, but it seems to occur if the device moves out of range and my handling logic attempts to reconnect on return.

I look a screenshot of tokio-console with the deadlock:

[image]

In particular, I already wrapped this connect call in a `tokio::time::timeout`, which you can see at the top of the screenshot.

- The one reference to `winrtable` is https://github.com/deviceplug/btleplug/blob/13a638ae0f81cb47077aae7cefbee29495681b4b/src/winrtble/peripheral.rs#L94
- I can't really see information on any of the other resources/tasks

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because the original Windows connection hang predates the current timeout-capable GATT operations and timeout fallback.

closed 2026-08-30 by qdot