deviceplug/btleplug #303: 0.10.5 fails to discover any devices on Ubuntu 22.04 LTS

ciniml, 2023-04-17:
**Describe the bug**

Actually, I think it is not a bug, but I want to submit this issue for informative.

Version 0.10.5 of this crate seems not working on Ubuntu 22.04 LTS, which uses Bluez 5.64.0.

The root cause is that `bluez-async` crate was updated from 0.60.0 to 0.71.0, which requires `Bonded` property for Device information, though the `Bonded` property was added from Bluez 5.65.0.

I've confirmed that 0.10.4 works well on my machine. So Ubuntu 22.04 LTS users should use version 0.10.4 instead of 0.10.5 until the Bluez for Ubuntu 22.04 is updated.

**Expected behavior**

**Actual behavior**

**Additional context**

qwandor, 2023-04-18:
Ah, sorry about that! It should be possible to fix this in `bluez-async`, I'll do that.

qwandor, 2023-04-18:
I've published a new version of `bluez-async` with the bugfix so if you update your `Cargo.lock` it should work.