deviceplug/btleplug #323: ASHA protocol feasibility

stefnotch, 2023-07-15:
**Help Request**

It's possible to stream audio to newer hearing aids with the ASHA protocol <https://source.android.com/docs/core/connect/bluetooth/asha>. However, that protocol is only supported by a few Android devices. It is not a thing on Windows nor Linux.

I personally have such hearing aids, and being able to connect them to my PC would be very valuable to me. 
So I'd be interested in writing an open source app that implements said protocol.

However, I'm an absolute novice when it comes to Bluetooth. Before attempting something that might be entirely too difficult for me, I wanted to ask if it should be possible to implement the ASHA protocol in an app with this library.

I'd also be more than willing to sponsor this project if answering that question takes a lot of research. I sadly can't sponsor it all that much, since I'm a student and not a billionaire.

qdot, 2023-07-15:
This is out of scope for this project. Another library could possibly use btleplug for the advertising/enumeration portions of the protocol, but the L2CAP and audio rendering portions are a huge lift that is well outside of what this library is made to do.

closed 2023-07-15 by qdot

stefnotch, 2023-07-17:
Thanks for the response. I ended up looking into L2CAP, and it seems to be far more difficult on Windows than I would have expected (seems to involve writing a driver). So yes, that absolutely seems out of scope.