deviceplug/btleplug #333: [Quesion] What is the associate type of `Manager` trait?

joe733, 2023-08-21:
Hi I'm new to rust. Since the discussions tab wasn't open, I'm asking a question here.

Here's my code:

```rust
use btleplug::api::{Central, Manager};
use std::{error::Error, any::TypeId};

pub async fn bluetooth() -> Result<(), Box<dyn Error>> {
    // Create a new manager.
    let manager = Manager::new().unwrap(); // type ERROR here !

    // Get the first Bluetooth adapter.
    let adapter = manager.adapters().await?.into_iter().nth(0).unwrap();

    // Create a central instance.
    let central = Central::new(adapter).await?;

    // Start scanning for devices.
    central.start_scan().await?;

    // list devices
    // how ?

    Ok(())
}
```

LSP complains:

```md
the value of the associated type `Adapter` (from trait `btleplug::api::Manager`) must be specified rustc(Click for full compiler diagnostic)
wireless.rs(8, 19): specify the associated type: `Manager<Adapter = Type>`
```

I'm supposed to annotate type as follows:

```rust
use btleplug::{api::{Central, Manager}, platform::Adapter};

...

let manager: Manager<Adapter = ?> = Manager::new().unwrap();
```

But what type goes in place of `?` ?

qwandor, 2023-08-21:
Rather than 

```rust
use btleplug::api::{Central, Manager};
```

try:

```rust
use btleplug::api::{Central, Manager as _};
use btleplug::platform::Manager;
```

`btleplug::api::Manager` is a trait, whereas `btleplug::platform::Manager` is a type implementing that trait. The compiler needs to know the concrete type you want to call `new` on.

joe733, 2023-08-21:
Thank you!

Also found the [docs](https://docs.rs/btleplug/latest/btleplug/) and [examples](https://github.com/deviceplug/btleplug/tree/master/examples).

closed 2023-08-21 by joe733