deviceplug/btleplug #334: Wrong peripheral name on macOS

Massimiliano-solutiontech, 2023-08-23:
**Describe the bug**
`properties.local_name` doesn't have the correct name because [macOS caches the names](https://developer.apple.com/forums/thread/19381#answers-title)

**Expected behavior**
Have the correct advertised name 

**Actual behavior**
Returning the cached named

**Additional context**
Below the Swift code working on iOS, iPadOS, but I didn't tested on macOS

https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/1518937-centralmanager

```swift
func centralManager(_ central: CBCentralManager, didDiscover peripheral: CBPeripheral, advertisementData: [String: Any], rssi RSSI: NSNumber) {
    guard let name = advertisementData["kCBAdvDataLocalName"] as? String else { return }
}
```

I have experience with Swift and iOS development, if you need help to implement that I'll be happy to do that

Massimiliano-solutiontech, 2023-09-17:
Any news on that?

gmallios, 2024-03-07:
I've implemented this and checked if the advertisement data contains the LocalName but no luck. 
Connecting to the device upon discovery seems to solve this issue.

yuyoyuppe, 2024-06-06:
Having the same issue, the fix by @Massimiliano-solutiontech has helped. In my case `kCBAdvDataLocalName` contained a far more descriptive name.

Specifically, my treadmill is displayed as "LifeSpan-TM-2000" when scanning on Windows with `btleplug`/nrfconnect on Android. However, on macOS 14.5 it's displayed as "ESP32".

I'm not sure what's the most appropriate fix would be, since exposing a new field in a `Peripheral` for all platforms would likely be an overkill, but I guess we can merge names on macOS If both fields are present?

qdot, 2024-06-06:
I *think* I'll be fine merging them, but I'll have to try them in my own library and see what it does.

yuyoyuppe, 2024-06-06:
@qdot rebased against `dev`.

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because current CoreBluetooth name handling preserves the advertisement name and has been corrected since this report.

closed 2026-08-30 by qdot