deviceplug/btleplug #342: BLE advertisements.

Frostie314159, 2023-10-08:
**Feature Description**

It would be helpful to register ble advertisements along with manufacturer data, since this is a feature supported by all platforms.

qdot, 2023-10-08:
Can you provide more context here? I'm not sure what you're asking for.

Frostie314159, 2023-10-08:
BLE allows you to advertise services. These can contain manufacturer data. As every platform supports sending BLE advertisements, I thought it would only make sense, if this crate exposed them as well.

qdot, 2023-10-09:
We already handle that, it's on the peripheral properties struct, as `manufacturer_data`.

Frostie314159, 2023-10-09:
I can't seem to find a way to advertise a service. The peripheral properties struct only handles receiving advertisements.

qdot, 2023-10-09:
That's out of the scope of this library. As the readme says, this is a host/central only library.

closed 2023-10-09 by qdot