deviceplug/btleplug #348: Windows "No such interface supported"

MGlolenstine, 2023-11-17:
**Describe the bug**
After implementing my [library](https://github.com/MGlolenstine/comm_handler) (on `bluetooth_support` branch) and testing it on Linux, where it worked, I decided to work on a Windows application that needs Bluetooth connection as well.

However, while the program connects to the device, gets the characteristic, writes to it, it fails to read from it.
<details>
<summary>Program log</summary>

```
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] services 4
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] characteristics Ok(3)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(1)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] characteristics Ok(3)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] characteristics Ok(3)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] characteristics Ok(3)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(0)
[2023-11-17T13:09:12Z DEBUG btleplug::winrtble::ble::device] descriptors Ok(1)
[2023-11-17T13:09:12Z DEBUG comm_handler::adapters::bluetooth] Writing [190, 201, 163, 251, 21, 152, 246, 118, 36, 57, 253, 29, 238, 49, 213, 42, 170, 199, 60, 116, 102, 138, 19, 243, 214, 242, 222, 182, 226, 13] to BT.
[2023-11-17T13:09:12Z ERROR comm_handler::framed_handler] An error occured while receiving data: Btleplug Error: Btleplug: Windows UWP threw error due to Protocol Error on read: IReference(IUnknown(0x26bd5b6de60)) Err(Error { code: HRESULT(0x80004002), message: "No such interface supported" }) Err(Error { code: HRESULT(0x80004002), message: "No such interface supported" })
```
</details>

I was able to track the issue to [this line](https://github.com/deviceplug/btleplug/blob/dev/src/winrtble/ble/characteristic.rs#L87), where it returns `IReference`, which contains `Err(Error { code: HRESULT(0x80004002), message: "No such interface supported" })` when opened with `IReference::GetString`.

**Expected behavior**
I'd expect Windows to be able to read and write from and to the BLE device.

**Actual behavior**
Windows seems to be able to scan, connect, get services, characteristics, but fails to read data from a notify characteristic.

**Additional context**
Add any other context about the problem here.

MGlolenstine, 2023-11-17:
It does seem to work fine on Windows 10.
The issue could be Windows 11 or just broken drivers.
I'll test if the drivers are the cause by reinstalling them on Monday and will report back.

qdot, 2023-11-19:
I just released a new version with an updated version of windows-rs. Not sure that'll help, but it's worth a shot.

MGlolenstine, 2023-11-20:
I've updated to the latest version and tested it, but I was still out of luck.

My issue was that I was reading from a characteristic that only implemented `NOTIFY`.
Now the error msg makes more sense, but it doesn't explain why it worked on some Windows devices and not on others (I tested it on another Windows 10, but it still failed).

Would it be possible to implement better error handling for scenarios like these?
(This time the error was on my side, but the error was confusing.)

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because the final discussion identifies this as an attempt to read a notify-only characteristic, not a btleplug discovery failure; the Windows error is a consequence of unsupported usage.

closed 2026-08-30 by qdot