deviceplug/btleplug #359: Droidplug has not been initialized (Android)

JosuGZ, 2023-12-27:
I'm getting this problem when running on Android: "Droidplug has not been initialized. Please initialize it with btleplug::platform::init().".

Should I have to call this myself? If that is the case an example on the README would be nice, although perhaps the problem is on how I'm (trying) to add the Java code to the project.

JosuGZ, 2023-12-27:
Ok, I made some progress, it seems that given `mut env: JNIEnv<'local>` I can add this (as long as I use the same version of the jni crate...):

```
#[cfg(target_os = "android")]
btleplug::platform::init(&env);
```

Now I'm getting this, which probably means I'm not loading the Java libraries properly...
```
JNI DETECTED ERROR IN APPLICATION: JNI NewStringUTF called with pending exception java.lang.ClassNotFoundException: Didn't find class "com.nonpolynomial.btleplug.android.impl.Adapter"
```

JosuGZ, 2023-12-27:
Adding this works:

```
#[cfg(target_os = "android")]
btleplug::platform::init(&env);
```

Setting gradle is tricky though.

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because the reporter confirmed that initializing btleplug with the Android environment resolves the reported error; the remaining class-loading issue belongs to the consuming integration.

closed 2026-08-30 by qdot