deviceplug/btleplug #364: Unable to scan on windows 10 tablet: "no such interface supported"

ChristianPavilonis, 2024-02-01:
**Describe the bug**
When calling `adapter.start_scan().await` I get this error: Error { code: HRESULT(0x800004002), message "no such interface supported" }

It only happens on the windows 10 tablet that I'm targeting other windows devices work.

**Expected behavior**
I expect it to start scanning like it does on every other device I test.

**Actual behavior**
The error above.

**Additional context**
The device is a faytech "10.1 Ruggedized Tablet PC"

It runs Windows 10 enterprise LTSC. 

Could it be the fact that it's an LTSC tablet that's is simply so out of date that it wont work with  the `windows` crate?

Might be out of scope for this crate but I'm not sure how to trouble shoot this. 🤔 

I made it panic so I could give you a stack trace.
```rust
thread 'tokio-runtime-worker' panicked at src\ble\discovery.rs:85:53:
called Result::unwrap() on an Err value: Other("Error { code: HRESULT(0x80004002), message: "No such interface supported" }")
stack backtrace:
0: std::panicking::begin_panic_handler
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library\std\src\panicking.rs:645
1: core::panicking::panic_fmt
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library\core\src\panicking.rs:72
2: core::result::unwrap_failed
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library\core\src\result.rs:1653
3: enum2$<core::result::Result<tuple$<>,enum2$btleplug::Error > >::unwrap<tuple$<>,enum2$btleplug::Error >
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\core\src\result.rs:1077
4: windows_app::ble::discovery::event_based_scan::async_fn$0
at .\src\ble\discovery.rs:85
5: windows_app::main::async_block$0::closure$0::async_block$0
at .\src\main.rs:82
6: tokio::runtime::task::core::impl$6::poll::closure$0<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\core.rs:328
7: tokio::loom::std::unsafe_cell::UnsafeCell<enum2$<tokio::runtime::task::core::Stage<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0> > > >::with_mut
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\loom\std\unsafe_cell.rs:16
8: tokio::runtime::task::core::Core<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >::poll<enum2$<windows_app::main::async_block$0::closure
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\core.rs:317
9: tokio::runtime::task::harness::poll_future::closure$0<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\harness.rs:485
10: core::panic::unwind_safe::impl$23::call_once<enum2$<core::task::poll::Poll<tuple$<> > >,tokio::runtime::task::harness::poll_future::closure_env$0<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arc<tokio::runtime::sched
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\core\src\panic\unwind_safe.rs:272
11: std::panicking::try::do_call<core::panic::unwind_safe::AssertUnwindSafe<tokio::runtime::task::harness::poll_future::closure_env$0<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arc<tokio::runtime::scheduler::multi_thre
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\std\src\panicking.rs:552
12: std::panicking::try::do_catch<core::panic::unwind_safe::AssertUnwindSafe<tokio::runtime::task::harness::impl$2::complete::closure_env$0<enum2$<sqlx_core::pool::inner::spawn_maintenance_tasks::async_block_env$1<sqlx_sqlite::database::Sqlite> >,alloc::sync:
13: std::panicking::try<enum2$<core::task::poll::Poll<tuple$<> > >,core::panic::unwind_safe::AssertUnwindSafe<tokio::runtime::task::harness::poll_future::closure_env$0<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arc<tok
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\std\src\panicking.rs:516
14: std::panic::catch_unwind<core::panic::unwind_safe::AssertUnwindSafe<tokio::runtime::task::harness::poll_future::closure_env$0<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arc<tokio::runtime::scheduler::multi_thread::
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\std\src\panic.rs:142
15: tokio::runtime::task::harness::poll_future<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\harness.rs:473
16: tokio::runtime::task::harness::Harness<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >::poll_inner<enum2$<windows_app::main::async_bloc
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\harness.rs:208
17: tokio::runtime::task::harness::Harness<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >::poll<enum2$<windows_app::main::async_block$0::c
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\harness.rs:153
18: tokio::runtime::task::raw::poll<enum2$<windows_app::main::async_block$0::closure$0::async_block_env$0>,alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\raw.rs:271
19: tokio::runtime::task::raw::RawTask::poll
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\raw.rs:201
20: tokio::runtime::task::LocalNotified<alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >::run<alloc::sync::Arctokio::runtime::scheduler::multi_thread::handle::Handle,alloc::alloc::Global >
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\mod.rs:416
21: tokio::runtime::scheduler::multi_thread::worker::impl$1::run_task::closure$0
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\scheduler\multi_thread\worker.rs:576
22: tokio::runtime::coop::with_budget
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\coop.rs:107
[... 41 lines cut ...]
44: std::panicking::try<enum2$<core::task::poll::Poll<tuple$<> > >,core::panic::unwind_safe::AssertUnwindSafe<tokio::runtime::task::harness::poll_future::closure_env$0<tokio::runtime::blocking::task::BlockingTask<tokio::runtime::scheduler::multi_thread::worke
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\std\src\panicking.rs:516
45: std::panic::catch_unwind<core::panic::unwind_safe::AssertUnwindSafe<tokio::runtime::task::harness::poll_future::closure_env$0<tokio::runtime::blocking::task::BlockingTasktokio::runtime::scheduler::multi_thread::worker::impl$0::launch::closure_env$0,toki
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\std\src\panic.rs:142
46: tokio::runtime::task::harness::poll_future<tokio::runtime::blocking::task::BlockingTasktokio::runtime::scheduler::multi_thread::worker::impl$0::launch::closure_env$0,tokio::runtime::blocking::schedule::BlockingSchedule>
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\harness.rs:473
47: tokio::runtime::task::harness::Harness<tokio::runtime::blocking::task::BlockingTasktokio::runtime::scheduler::multi_thread::worker::impl$0::launch::closure_env$0,tokio::runtime::blocking::schedule::BlockingSchedule>::poll_inner<tokio::runtime::blocking:
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\task\harness.rs:208
48: tokio::runtime::task::harness::Harness<tokio::runtime::blocking::task::BlockingTasktokio::runtime::scheduler::multi_thread::worker::impl$0::launch::closure_env$0,tokio::runtime::blocking::schedule::Block [... cut ...]
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\blocking\pool.rs:159
53: tokio::runtime::blocking::pool::Inner::run
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\blocking\pool.rs:513
54: tokio::runtime::blocking::pool::impl$6::spawn_thread::closure$0
at C:\Users\fay.cargo\registry\src\index.crates.io-6f17d22bba15001f\tokio-1.35.1\src\runtime\blocking\pool.rs:471
55: core::hint::black_box
at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112\library\core\src\hint.rs:286
```

qdot, 2024-02-01:
What's the build version of that windows 10 release?

ChristianPavilonis, 2024-02-01:
> What's the build version of that windows 10 release?

Version: 1809
OS build 17763.5329

ChristianPavilonis, 2024-02-01:
Update: I removed that line in my fork and I'm able to scan.

qdot, 2024-02-01:
Huh, interesting. I wonder if we can do some sort of existence check around that.

ChristianPavilonis, 2024-02-02:
Yeah up to you if you want to support a 6 year old version of windows. ¯\_(ツ)_/¯

wuwbobo2021, 2024-09-09:
I've got the same error when testing on Windows 10 LTSC 2019, which is supported by Microsoft until 2029 (longer than LTSC 2021). Version 1809, 17763.5936.

Would you please describe OS version requirements in the documentation before fixing this issue?

"btleplug is a Bluetooth Low Energy (BLE) central module library for Rust. It currently supports Windows 10, macOS (and possibly iOS) and Linux (BlueZ)."

Does it support only Windows 10 2004 and later, but not Windows 11?I've got the same error when testing on Windows 10 LTSC 2019, which is supported by Microsoft until 2029 (longer than LTSC 2021). Version 1809, 17763.5936.

Would you please describe OS version requirements in the documentation before fixing this issue?

"btleplug is a Bluetooth Low Energy (BLE) central module library for Rust. It currently supports Windows 10, macOS (and possibly iOS) and Linux (BlueZ)."

Does it support only Windows 10 2004 and later, but not Windows 11?

wuwbobo2021, 2024-09-10:
Workaround in my app: in `btleplug-0.11.5\src\winrtble\ble\watcher.rs`, line 54, change `self.watcher.SetAllowExtendedAdvertisements(true)?;` to `let _ = self.watcher.SetAllowExtendedAdvertisements(true);`.

<https://github.com/wuwbobo2021/rtl8762c-ble-uart/blob/main/host-term/README.md>

Another interesting problem: when bluetooth is disabled on Windows 10, the adapter is still available in `btleplug` (`start_scan()` can be called). And I don't know how to get the ATT_MTU value of the connection (maybe not an issue of `btleplug`).

wuwbobo2021, 2024-12-22:
I'm switching to [bluest](https://docs.rs/bluest/0.6.7/bluest/) because the only maintainer of `btleplug` is a sex toy maker from "Nonpolynomial LLC". It's unlikely for me to use this crate anymore.

I has been annoyed by the huge horrible icon in <https://crates.io/crates/buttplug>, which I occasionally discovered before checking the meaning of "butt" in the dictionary. After that I decided to report that crate on crates.io, however the reporting web page doesn't seems to be responding (I make deal with the crates.io bug later).

<https://www.rust-lang.org/policies/code-of-conduct>
<https://github.com/stumpsyn/policies/blob/master/citizen_code_of_conduct.md> (4. Unacceptable Behavior)

Off-topic: As a non-Christian from China, I was suprised to see the guy named "Christian" talking to you friendly.

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because Windows 10 build 1809 does not support the extended-advertising API used by this path; the current code already ignores that unavailable capability, making the remaining limitation an OS/API constraint.

closed 2026-08-30 by qdot

wuwbobo, 2026-10-18:
@qDot I am really, really sorry for my previous offensive comments. I deleted my previous account so that I can report and even fix bugs of `btleplug` in the future, at least for desktop platforms. My previous misunderstanding about the crate name is explained in <https://github.com/kevinmehall/nusb/pull/150#issuecomment-6112259809>.