deviceplug/btleplug #368: Missing Values For Device Address and TX Power

ztroop, 2024-03-02:
**Describe the bug**
I'm writing a program that uses this library to scan devices. I'm able to scan and detect devices, but I'm getting `00:00:00:00:00:00` for `address` and `None` for `tx_power_level` consistently across 15+ devices. 

```rs
    let manager = Manager::new().await?;
    let adapters = manager.adapters().await?;
    let central = adapters.into_iter().next().ok_or("No adapters found")?;

    central.start_scan(ScanFilter::default()).await?;
    let mut events = central.events().await?;

    let mut devices_info = Vec::new(); // Accumulate device information here

    while let Some(event) = events.next().await {
        if let CentralEvent::DeviceDiscovered(id) = event {
            if let Ok(device) = central.peripheral(&id).await {
                let properties = device
                    .properties()
                    .await?
                    .unwrap_or(PeripheralProperties::default());
                let name = properties
                    .local_name
                    .unwrap_or_else(|| "Unknown".to_string());
                let tx_power = properties
                    .tx_power_level
                    .map_or_else(|| "N/A".to_string(), |tx| tx.to_string());
                let address = properties.address.to_string();
                let rssi = properties
                    .rssi
                    .map_or_else(|| "N/A".to_string(), |rssi| rssi.to_string());
```

**Expected behavior**
Expecting data to be provided by `properties.address` and `properties.tx_power_level`.

**Actual behavior**
Consistent lack of data.

**Additional context**
Using version `0.11.5`.

qwandor, 2024-03-02:
Is this on Mac OS? Some platforms don't provide this information, in which case there's not much we can do.

ztroop, 2024-03-02:
Yes, this is a Mac Air (M2). Is there a matrix of what platforms are supported and offer what type of functionality?

ztroop, 2024-03-03:
Perhaps this is the case for `tx_power_level` but `address` should show something other than 00:00:00:00:00:00, as other BTLE libraries are able to retrieve this information. I suspect there might be going on there.

qdot, 2024-03-03:
It can't show anything, because CoreBluetooth does not allow access to device addresses.

ztroop, 2024-03-04:
So I've been experimenting with different versions originally thinking it was a bug. When I downgrade to `0.8`, I see the addresses without issue:

[image]

Is this because we're using a different method in the older version to get the `address`?

clouds56, 2024-03-06:
> So I've been experimenting with different versions originally thinking it was a bug. When I downgrade to `0.8`, I see the `address` without issue:
> 
> <img alt="Screenshot 2024-03-03 at 11 08 39 PM" width="889" src="https://private-user-images.githubusercontent.com/5018339/309625410-5dcb75bb-33af-4b58-9e0a-cd14046be73f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk3NDEwMTYsIm5iZiI6MTcwOTc0MDcxNiwicGF0aCI6Ii81MDE4MzM5LzMwOTYyNTQxMC01ZGNiNzViYi0zM2FmLTRiNTgtOWUwYS1jZDE0MDQ2YmU3M2YucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMDZUMTU1ODM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDBmNjA5MjUzZjNjYjY1Y2Y0NmVhZjgyYjA4ZTk5NDExMGNiYzVkMTkxMjViNDM3YzU3OTllMGQwOTViYzgzNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.syc9fbF6mFRpcb24JNnV5PztWuZaG72224vXBv3EEtk">
> Is this because we're using a different method in the older version to get the `address`?

It appears the `address` calculation in v0.8 differs from the current method. In v0.8, `BDAddr` is solely derived from the `uuid` or `PeripheralId`, which doesn't correspond to the actual address.
https://github.com/deviceplug/btleplug/blob/e7eb1c875a317e4c86a13e1afe9173ff1b07bdbd/src/corebluetooth/adapter.rs#L22-L25

ztroop, 2024-03-07:
I think that's expected though.
> The CoreBluetooth APIs in iOS do not give you any direct access to a peripheral's real MAC address; instead, each new peripheral seen is assigned a 128-bit UUID and used to reference that peripheral object permanently. [See Reference](https://community.silabs.com/s/article/x-how-to-get-a-ble-peripheral-mac-address-with-ios-and-corebluetooth?language=en_US)

Given the lack of a consistent method to retrieve the actual MAC address, is it still possible to access the `uuid` as in previous versions? After reviewing the code, it remains unclear to me whether this information is still accessible.

I believe for my purposes the `uuid` would be a satisfactory replacement - even if that would appear different from another scanning device.

ztroop, 2024-03-07:
In `corebluetooth/peripheral.rs`:

```rs
pub struct Peripheral {
    shared: Arc<Shared>,
}

struct Shared {
    notifications_channel: broadcast::Sender<ValueNotification>,
    manager: Weak<AdapterManager<Peripheral>>,
    uuid: Uuid,
    ...
```

I think the `uuid` mentioned in my previous post is found in `Shared`. However, that's private and not accessible.

ztroop, 2024-03-09:
> PeripheralId is implemented with Display traits on all platforms, and will encase the platform specific identifier for the device, be it BTAddr or UUID. You can use that across platforms for identification.

I'm good with this. Closing.

closed 2024-03-09 by ztroop