deviceplug/btleplug #370: Scan filtering broken on Windows 11

markATavail, 2024-03-07:
As a data point, I believe scan filtering has stayed working on Mac since well before this patch.  On Windows 11, running the same application as on Mac (Kotlin connecting to a Rust library including btleplug), it used to work correctly when one advertisement id was in the filter, but always reported no advertisements if there were two or more ids in the filter list.  Now it presents an empty list of advertisements if even one id is in the filter list.  The related pull request #313 looks straightforward to me, but maybe there's something broken with filtering in the underlying library.

Expected: Expect scan filtering to work on Windows 11.

Actual: The advertisement list is always empty on Windows 11, if any id is included in the `ScanFilter { services }` vec.

Additional context: This works on Mac when presented with zero, one, or two services in the ScanFilter, correctly filtering out non-matching services.  I don't have access to Windows 10 for testing.

HoshinoSuzumi, 2024-07-15:
I'm having the same problem.

mgibbs189, 2025-04-10:
Hey @markATavail @HoshinoSuzumi have you had any luck with this issue?

markATavail, 2025-04-14:
I'm no longer working on that project, but as of a few months ago it was still broken.  The workaround was either to ignore the filter and sort through a very long list of BLE devices, or to run two separate queries with singular filters.  Not ideal, but our particular case involved devices with one of exactly two service ids.

mgibbs189, 2025-04-15:
@markATavail good to know, thanks!

NielsMeereboer, 2025-10-26:
From my investigation I conclude that it is not supported my microsoft.

On Windows 10/11, the filter does not behave as an OR filter — it behaves like an AND filter. Only advertisements containing all of the specified service UUIDs will pass through. So by adding more uuids, your filter becomes more strict. You may want to include a callback yourself to apply filtering.

In more confusing words: 
https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementfilter?view=winrt-26100

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because the current Windows watcher performs software scan-filter matching instead of relying on the problematic OS-level filter behaviour described in this report.

closed 2026-08-30 by qdot