deviceplug/btleplug #374: compilation error when compiling to arm

alexou2, 2024-03-21:
**Describe the bug**
i get a pkg config error when i try to compile my code to `armv7-unknown-linux-gnueabihf`

**Expected behavior**
the code should compile without an errors

**Actual behavior**
when i compile my code for `armv7-unknown-linux-gnueabihf`, get this error: `pkg_config failed: pkg-config has not been configured to support cross-compilation.`. 

**Additional context**
I am compiling from x86 linux

qwandor, 2024-03-22:
Have you set up `pkg-config` for cross compilation? You'll need `libdbus` to link against, at least. [cross](https://crates.io/crates/cross) is one way to do this, you'll want something like [this Dockerfile](https://github.com/alsuren/mijia-homie/blob/master/docker/Dockerfile.debian-buster-armv7).

alexou2, 2024-03-22:
I'm not familiar with rust cross compilation. I tried to set up pkg config, but It didn't change anything. I can compile my code for arm if I don't use btleplug

alexou2, 2024-03-22:
I was able to set up pkg-config for cross compilation, but now i get this linker error: `/usr/lib/gcc-cross/arm-linux-gnueabihf/11/../../../../arm-linux-gnueabihf/bin/ld: cannot find -ldbus-1: No such file or directory
          collect2: error: ld returned 1 exit status`

alexou2, 2024-03-23:
now I get this error when compiling with cross `The system library `dbus-1` required by crate `libdbus-sys` was not found.`

qwandor, 2024-03-24:
Did you install libdbus-1-dev in the Docker container for cross, using the Dockerfile I linked or otherwise?

qdot, 2026-08-31:
I'm gonna close this out because it's 2+ years old and seems to be an environment issue. Please reopen or refile if another issue arises.

closed 2026-08-31 by qdot