deviceplug/btleplug #375: Android StartScan call can throw unhandled NullPointerException

qdot, 2024-03-26:
Finally got a logcat from someone seeing crashes on Android when trying to start a bluetooth scan.

```
03-25 09:45:12.943 27456 27563 F ntiface_centra: thread.cc:2477] No pending exception expected: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.startScan(java.util.List, android.bluetooth.le.ScanSettings, android.bluetooth.le.ScanCallback)' on a null object reference
03-25 09:45:12.943 27456 27563 F ntiface_centra: thread.cc:2477]   at void com.nonpolynomial.btleplug.android.impl.Adapter.startScan(com.nonpolynomial.btleplug.android.impl.ScanFilter) (Adapter.java:9)
03-25 09:45:12.943 27456 27563 F ntiface_centra: thread.cc:2477] 
0
```

It looks like something in this line is causing this issue:

```java
BluetoothAdapter.getDefaultAdapter().getBluetoothLeScanner().startScan(filters, settings, this.callback);
```

Most likely, getDefaultAdapter() is returning null here for some reason, but I'm not sure why. Unfortunately, since our Android code runs without a context, we can't check for system service status before trying this either.

At minimum, we should at least check for null on both the adapter and scanner objects, and throw with a relevant message. Ideally though, it'd be nice to figure out why we're not able to fetch an adapter here.

Gibbz, 2024-04-19:
Sorry to hijack the thread. How do you setup the android permissions?
I've got the startscan working on desktop, but android nothing happens for me. And android never asked for Bluetooth permission.

qdot, 2024-04-19:
@Gibbz Required permissions are listed in the app manifest: https://github.com/intiface/intiface-central/blob/main/android/app/src/main/AndroidManifest.xml

And we ask for them on startup: https://github.com/intiface/intiface-central/blob/main/lib/intiface_central_app.dart#L178

wuwbobo2021, 2024-10-19:
"our Android code runs without a context"

Is this crate helpful? <https://docs.rs/ndk-context/latest/ndk_context/struct.AndroidContext.html#method.context>

It might work with `j4rs` as well: <https://github.com/astonbitecode/j4rs/issues/121>

qdot, 2024-10-19:
Yeah I've got something filed to move to j4rs, it's just going to take more work than I've had time for lately. https://github.com/deviceplug/btleplug/issues/376

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because the Android scan path now checks for both a missing Bluetooth adapter and a missing scanner before starting a scan.

closed 2026-08-30 by qdot