deviceplug/btleplug #377: Converting string to PeripheralId

alexou2, 2024-03-28:
**Feature Description**

I would need to convert a string to a PeripheralId in order to connect to the device. I could see something like this:
`pub async fn connect_to_device(device_id: &str) -> Result<(), Error> {
    println!("{}", device_id);
    let manager:Manager = Manager::new().await?;
    let central:Adapter = get_central(&manager).await?;
    let device:DeviceId = DeviceId::from(device_id);
    let peripheral = central.peripheral(device_id).await?;
    peripheral.connect().await?;
    println!("device connected");
    Ok(())
}`

qwandor, 2024-03-28:
Where would you get the peripheral ID string from? The peripheral ID is quite different on each platform, and is generally best treated as an opaque token.

alexou2, 2024-03-29:
What i want to do is storing something like the bluetooth address of a device as a string and convert it to a peripheralId in order to connect to the device

qwandor, 2024-04-04:
`PeripheralId` doesn't necessarily represent the address of the peripheral, it is a platform-dependent opaque identifier and you shouldn't rely on its format. If you want to search for a device with a particular address you can iterate over all discovered devices and call `Peripheral::address` to get the address. Note that Mac OS won't give you addresses at all though.

closed 2024-04-04 by qwandor

Yongle-Fu, 2024-08-23:
```rust
pub fn to_peripheral_id(id: *const c_char) -> PeripheralId {
    let id_str = unsafe { CStr::from_ptr(id) };
    let id_str = id_str.to_str().expect("Invalid UTF-8 sequence");
    info!("to_peripheral_id, id_str: {:?}", id_str);

    #[cfg(any(target_os = "macos", target_os = "ios"))]
    return PeripheralId::from(Uuid::parse_str(id_str).unwrap());

    #[cfg(target_os = "android")]
    return PeripheralId(BDAddr::from_str_delim(&id_str).unwrap());

    #[cfg(target_os = "windows")]
    PeripheralId::from(BDAddr::from_str_delim(&id_str).unwrap())
}
```

why not provide public method

droidplug/peripheral.rs:42:25
   |
42 | pub struct PeripheralId(pub(super) BDAddr);
   |                         ^^^^^^^^^^^^^^^^^ private field