deviceplug/btleplug #383: v0.5 Adapter::connect() fails with EINVAL

sooda, 2024-05-11:
**Describe the bug**

On a system with a sufficiently recent Linux kernel, Adapter::connect() in btleplug 0.5 (that still uses sockets which may be preferable to some less frequently updated users of this library, especially considering the async overhaul that comes with later versions) now fails, returning Error::Other("EINVAL: Invalid argument"), originating from set_socket_filter().

**Expected behavior**

Successful connected adapter returned

**Actual behavior**

"EINVAL: Invalid argument" returned

**Additional context**

Normally Linux does not break userspace but Luiz Augusto von Dentz recently did a Mauro while applying a reasonable fix:

https://lore.kernel.org/all/20240410191610.4156653-1-luiz.dentz@gmail.com/

Whether or not that really broke userspace or if this particular userspace (that's us) was already broken to begin with would be debatable (this did work fine before, after all), as there would be no problem if the proper sizeof(struct hci_filter) was used:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/lib/hci.h

But btleplug is not C and builds the structure field by field, missing padding at the end of the struct. This has not been a cause of runtime problems in practice as the padding bits are unspecified anyway.

sooda, 2024-05-11:
Proposed fix in https://github.com/deviceplug/btleplug/pull/384

Another "fix" for users would be to use a later version of this library, but the async overhaul is a bit much. 0.6 that got rid of these sockets in favor of dbus broke some features (like active() and filter_duplicates()) so a simpler update from 0.5 to 0.6 is not an option.

qdot, 2024-05-11:
We don't have the resources to maintain multiple old versions. Feel free to fork at v0.5 if you need a version kept up, but I have no plans on maintenance for a 3.5 year old branch on what is now a completely different library.

closed 2024-10-05 by qdot