deviceplug/btleplug #392: OpenHarmony support.

laxian, 2024-07-16:
**Feature Description**

**HarmonyOS NEXT** is coming.  I want to use `btleplug` in my HarmonyOS App. 

I want to use this library in harmonyOS project, could you consider supporting it? 
Or if there's anyone with the same interests.

qdot, 2024-07-16:
Seeing as this is literally the first I've ever heard of HarmonyOS: No, no interest in supporting it at this time. 

I'll leave this open, but I doubt we'd even take a PR at this point, because support for a whole platform is too much of a support burden for too small of a user base (we've never had any requests for this before).

laxian, 2024-07-19:
> Seeing as this is literally the first I've ever heard of HarmonyOS: No, no interest in supporting it at this time.
>
> I'll leave this open, but I doubt we'd even take a PR at this point, because [...]

[OpenHarmony](https://github.com/openharmony)   is NOT _to small of a user base_

mdddj, 2024-07-23:
可以试试ohos-rs来编译它

laxian, 2024-07-29:
> 可以试试ohos-rs来编译它

我目前已经在使用ohos-rs在实现了

wuwbobo2021, 2025-02-20:
You don't have to choose `btleplug` which is originally a support crate for the authors money maker `buttplug`. There are some other alternatives like `bluest`, you can add support for one of them, otherwise just create a separate library.

Put the big icon of `buttplug` and the logo of `HarmonyOS` together and look at this insult... Is that really what you want?! Do you really agree that `HarmonyOS` can be an ***hole?

laxian, 2025-02-21:
> You don't have to choose `btleplug` which is originally a support crate for the authors money maker `buttplug`. There are some other alternatives like `bluest`, you can add support for one of them, [...]
bluest doesn’t support android `Android support is planned. `

qdot, 2025-02-21:
Ok, having at least taken a look at the oh bluetooth layer, it doesn't look... *that* out of the question? Like there's C exposure of the GATT API and all, though I admittedly haven't looked at how ohos-rs handles those exports yet. I was mainly worried we'd be up against more java or something.

That said, I'm sitting here in a country that has little OH traction that I'm aware of (and I'm an embedded dev by trade so I feel like I at least have some idea of what's popular :) ), with no *clue* how much of a userbase there is for this elsewhere in the world. The gitee project has some impressive numbers (the github fork, not so much), but what's the rust ecosystem like on openharmony? As I see it, we currently have 2 users on this issue asking us to add a whole new platform to our library, which means maintenance, upkeep, etc... That's a lot to be asking if 1-2 people just want to use this.

qdot, 2025-02-21:
(Also, somewhat offtopic: What's the situation with flutter on openharmony? Does it work ok? :) )

wuwbobo, 2026-11-15:
@qdot Sorry, I would remind you of deleting <https://github.com/deviceplug/btleplug/issues/392#issuecomment-2673428300>. I know that the referenced comment in it makes absolutely no sense, and it is just annoying. I don't want that to have any negative influence to this project anymore. My motivation is described in <https://github.com/deviceplug/btleplug/issues/364#issuecomment-6244582060>.

Also, this comment itself is to be deleted later.