deviceplug/btleplug #394: Cannot run on M1 series chip macos

mdddj, 2024-07-23:
I tried to write a Flutter full platform package using flutter_rust_bridge, but encountered some errors
```dart
$ flutter run -d macos              
Launching lib/main.dart on macOS in debug mode...
Running pod install...                                           1,337ms
--- xcodebuild: WARNING: Using the first of multiple matching destinations:
{ platform:macOS, arch:arm64, id:00006000-0006395034D8801E, name:My Mac }
{ platform:macOS, arch:x86_64, id:00006000-0006395034D8801E, name:My Mac }
Undefined symbols for architecture arm64:
  "_CBAdvertisementDataManufacturerDataKey", referenced from:
      btleplug::corebluetooth::central_delegate::CentralDelegate::delegate_centralmanager_diddiscoverperipheral_advertisementdata_rssi::hdba26eb3f8ee3663 in libblue_super.a[arm64][199](btleplug-448251af3f53716a.btleplug.27c43d2c259b7feb-cgu.02.rcgu.o)
  "_CBAdvertisementDataServiceDataKey", referenced from:
      btleplug::corebluetooth::central_delegate::CentralDelegate::delegate_centralmanager_diddiscoverperipheral_advertisementdata_rssi::hdba26eb3f8ee3663 in libblue_super.a[arm64][199](btleplug-448251af3f53716a.btleplug.27c43d2c259b7feb-cgu.02.rcgu.o)
  "_CBAdvertisementDataServiceUUIDsKey", referenced from:
      btleplug::corebluetooth::central_delegate::CentralDelegate::delegate_centralmanager_diddiscoverperipheral_advertisementdata_rssi::hdba26eb3f8ee3663 in libblue_super.a[arm64][199](btleplug-448251af3f53716a.btleplug.27c43d2c259b7feb-cgu.02.rcgu.o)
  "_CBCentralManagerScanOptionAllowDuplicatesKey", referenced from:
      btleplug::corebluetooth::internal::CoreBluetoothInternal::start_discovery::h0bff35040dd318f1 in libblue_super.a[arm64][206](btleplug-448251af3f53716a.btleplug.27c43d2c259b7feb-cgu.09.rcgu.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
warning: Run script build phase 'Run Script' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'Flutter Assemble' from project 'Runner')
** BUILD FAILED **

Building macOS application...                                           
Error: Build process failed
```

qdot, 2024-07-31:
I vaguely remember running into this when building another flutter app that uses this library. I'll have to dig through my project and see if I can find what setting I used to fix it, because I'm building fine on m1 now.

laxian, 2024-11-03:
I'm building fine on m1 too.
Maybe you should add CoreBluetooth lib in xcode

mdddj, 2024-11-05:
> I'm building fine on m1 too. Maybe you should add CoreBluetooth lib in xcode
> 
> @mdddj

After adding it, the problem was solved. Thanks

closed 2024-11-06 by qdot