deviceplug/btleplug #398: Function `Central::peripherals` on android doesn't return any peripherals

bWanShiTong, 2024-10-02:
**Describe the bug**
Function `Central::peripherals` on android doesn't return any peripherals

**Expected behavior**
To work just as on linux, and return devices near by

**Actual behavior**
Function `Central::peripherals` on android doesn't return any peripherals

**Additional context**
I am using btleplug on android with tauri and `tauri-plugin-btleplug`

bWanShiTong, 2024-10-04:
This neither works on latest release on cargo, or on dev branch

qdot, 2024-10-05:
This works fine. I'm currently shipping an app with 10ks of downloads on android with it, with no reported issues. The problem may be permissions in your app, or something with the tauri-plugin-btleplug package that we do not maintain. We can't really help with either of those.

closed 2024-10-05 by qdot

bWanShiTong, 2024-10-05:
You were right about permissions, and it now it works, but if ble is turned off and `start_scan` function is called everything crashes:

These are the logs
```
[31m10-05 20:52:06.806  3025  3319 F .blescanner.app: java_vm_ext.cc:591] JNI DETECTED ERROR IN APPLICATION: JNI FindClass called with pending exception java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.startScan(java.util.List, android.bluetooth.le.ScanSettings, android.bluetooth.le.ScanCallback)' on a null object reference[0m
[31m10-05 20:52:06.806  3025  3319 F .blescanner.app: java_vm_ext.cc:591]   at void com.nonpolynomial.btleplug.android.impl.Adapter.startScan(com.nonpolynomial.btleplug.android.impl.ScanFilter) (Adapter.java:34)[0m
[31m10-05 20:52:06.806  3025  3319 F .blescanner.app: java_vm_ext.cc:591] [0m
[31m10-05 20:52:06.806  3025  3319 F .blescanner.app: java_vm_ext.cc:591]     in call to FindClass[0m
[31m10-05 20:52:07.457  3025  3319 F .blescanner.app: runtime.cc:691] Runtime aborting...[0m
[31m10-05 20:52:07.457  3025  3319 F .blescanner.app: runtime.cc:691] Dumping all threads without mutator lock held[0m
[31m10-05 20:52:07.457  3025  3319 F .blescanner.app: runtime.cc:691] All threads:[0m
```

Is there a way to handle this from rust

qdot, 2024-10-05:
Ok so *that* is actually a bug. We've already got it filed.

https://github.com/deviceplug/btleplug/issues/375