deviceplug/btleplug #412: “Software caused connection abort”

laxian, 2024-12-10:
**Describe the bug**
My code run on mobile And macos. But when I compile it on Ubuntu, it doesn't work.
I Get this error after each successful connection. Why?

laxian, 2024-12-11:
My code works fine on all mobile platforms and macos, and scans fine when I try to run it on Ubuntu, but this error occurs every time I connect. I'm running in docker. Is that relevant?

laxian, 2024-12-11:
[image]

qdot, 2024-12-21:
Possibly. Hardware access in docker can be *very* weird. Are you seeing connections on the device side?

laxian, 2025-02-19:
> Possibly. Hardware access in docker can be _very_ weird. Are you seeing connections on the device side?

Yes, there was a connection for a moment. 
I tested a non-docker environment on Ubuntu and had the same error.

pgschinghum, 2025-03-04:
I have got exactly the same issue and the application is not running inside docker, so the two are unrelated.

Weirdly however, one in 50 (or so) connection attempts works.