deviceplug/btleplug #418: Android setup documentation

Gibbz, 2024-12-24:
I'm working on adding this to my android project. The project is using xbuild and slint, which uses and android activity crate ( https://github.com/slint-ui/slint/tree/master/internal/backends/android-activity ).   

Do I need to modify the slint source code to add the btleplug stuff?

Ive completed the steps below:
```
<cd to Projects folder>
git clone https://github.com/deviceplug/jni-utils-rs.git
cd jni-utils-rs
cargo build --features=build-java-support
jar files located in /target/debug/java/libs

cd ..
git clone https://github.com/deviceplug/btleplug.git
cd btleplug/src/droidplug/java
edit build.gradle, comment and replace:
dependencies {
    implementation files('/home/bronson/Projects/jni-utils-rs/target/debug/java/libs/jni-utils-0.1.1-SNAPSHOT.jar')
}
./gradlew build 
btleplug/src/droidplug/java/build/outputs/aar/
```

This gets step 1 complete, and most of step 2... But how do i do step 2? ```have the Java portion of your android app point to that as a local implementation``` when slint takes care of this all for me?

```
A quick overview of the build process:

1  For java, you will need the java portion of [jni-utils-rs](https://github.com/deviceplug/jni-utils-rs) available either in a Maven repository or locally (if locally, you'll need to check out btleplug and change the gradle file).
2.  Either build the java portion of btleplug, in the src/droidplug/java directory, using the included gradle files, and them to a Maven repo, or have the Java portion of your android app point to that as a local implementation.
3.  For Rust, the build should go as normal, though we recommend using cargo-ndk to build. Output the jniLibs and make sure they end up in the right place in your app.
```

qdot, 2024-12-26:
This assumes I know anything about android build systems, which I absolutely do not, so I can't really help on that side.

I've handled this in my flutter app by just leaving the .aar in the app repo: https://github.com/intiface/intiface-central/tree/main/android/libs

So you may be able to just pull that .aar file, put it in your libs directory for your app, and call it done without needing to build everything yourself.

Gibbz, 2024-12-27:
I see your library is referenced in the gradle files. Does your Xbuild command build the gradle files too?

I've added gradle: true to my yaml. But the gradle files still fail to be build by X build. 

I'm wondering if there's a step I'm missing....

qdot, 2024-12-27:
I have no idea what xbuild is.

Gibbz, 2024-12-28:
which build tools are you using? ndk + gradle or cargo-apk?

qdot, 2024-12-28:
For the droidplug stuff I think I used ndk + gradle. IT's been a *long* time since I've built it.