deviceplug/btleplug #421: Consider using inMemoryDexClassLoader to load Java classes

wszxl516, 2025-01-24:
**Feature Description**

[like this](https://github.com/slint-ui/slint/blob/master/internal%2Fbackends%2Fandroid-activity%2Fjavahelper.rs)

Gradle is difficult to use for people who don't know Java.
I have successfully used dexloader to load bluey's java code. It's really simple compared to using gradle.

yehun, 2025-07-26:
I migrated [btleplug](https://github.com/deviceplug/btleplug) and [jni-utils](https://github.com/deviceplug/jni-utils-rs) locally, and then changed env.find_class, which supports Android, to classloader. I successfully compiled the Android apk on slint and ran it successfully.

reformc, 2025-12-30:
How did you manage to do this? I'm looking to achieve the same — thanks!