deviceplug/btleplug #432: Get Adapter Mac Address

DanielHabenicht, 2025-05-18:
**Feature Description**

To know which adapter is selected / select the right adapter it would be great to be able to access its mac address. 
As [`adapter_info` should not be relied on](https://github.com/deviceplug/btleplug/blob/c227cc1adf885e143217f274758968afc02490fb/src/api/mod.rs#L372-L376). 

Example: 
```rust
impl Central for Adapter {
    async fn adapter_mac(&self) -> Result<BDAddr> {
        // Get actual adapter address...
        Ok(BDAddr::from("B8:27:EB:7C:51:30"))
    }
}
```

qdot, 2025-05-18:
Welp, yet another "not gonna work on all our platforms" issue. Android just gives everyone 02:00:00:00:00:00 for security reasons (this is a fingerprintable address after all). I'm guessing iOS just straight up bans access to it, wouldn't be surprised if macOS is the same. So we've gotta figure out what that's going to look like on the two platforms you didn't provide solutions for before we can do this.

qdot, 2026-08-30:
> *This was generated by AI during triage.*

Closing this issue because the requested adapter-address API is implemented; macOS correctly reports that the platform does not expose a hardware address.

closed 2026-08-30 by qdot